Nevvman18/rs41-nfw #1: Failed to erase flash memory

videobuff, 2024-08-27:
I'm stuck for the whole evening because the STM32Cube wont erase the flash memory. And it wont write the firmware becasuse i have to tell the start address in Cube. 
Btw i can programm the older Vaisala's with the Cube.
As to flash protection:
The OB setting for RDP is AA.
Any help is much appreciated, read much on google and AI but so far no suc6

Erik, PA0ESH

Log STM32Cube programmer

23:44:48 : STM32CubeProgrammer API v2.16.0 | MacOS-64Bits 
23:46:02 : UR connection mode is defined with the HWrst reset mode
23:46:02 : ST-LINK SN  : 48FF6F065088515307380787
23:46:02 : ST-LINK FW  : V2J43S7
23:46:02 : Board       : --
23:46:02 : Voltage     : 3.25V
23:46:02 : SWD freq    : 4000 KHz
23:46:02 : Connect mode: Normal
23:46:02 : Reset mode  : Software reset
23:46:02 : Device ID   : 0x464
23:46:02 : Revision ID : Rev A
23:46:02 : Debug in Low Power mode enabled.
23:46:03 : UPLOADING OPTION BYTES DATA ...
23:46:03 :   Bank          : 0x00
23:46:03 :   Address       : 0x40022020
23:46:03 :   Size          : 20 Bytes
23:46:03 : UPLOADING ...
23:46:03 :   Size          : 1024 Bytes
23:46:03 :   Address       : 0x8000000
23:46:03 : Read progress:
23:46:03 : Data read successfully
23:46:03 : Time elapsed during the read operation is: 00:00:00.026
23:49:06 : MASS ERASE ... 
23:49:06 : Error: Mass erase operation failed.Please verify flash protection
23:49:12 : Memory Programming ...
23:49:12 : Opening and parsing file: RS41ng.elf
23:49:13 :   File          : RS41ng.elf
23:49:13 :   Size          : 51.54 KB 
23:49:13 :   Address       : 0x08000000 
23:49:13 : Erasing memory corresponding to segment 0:
23:49:13 : Erasing internal memory sectors [0 25]
23:49:13 : Error: failed to erase memory
23:49:13 : Error: failed to erase memory

[image]

Nevvman18, 2024-08-27:
Weird, I will look closely tomorrow morning at this and try flashing a factory new one to compare.

Nevvman18, 2024-08-27:
[image]
These are my board settings, they look like this above. Please try to set these settings in your environment and tell me if it has worked. I probably forgot to mention this part of the process in the manual, sorry.
ALso make sure that you flash correct file at the correct start address.

videobuff, 2024-08-28:
Good Morning. Following the settings above, the full chip erase succeeded (message: Mass erase succesfully achieved). I had to enter a start adress before flashing succeeded. Thanks for the quick help

[image]

Nevvman18, 2024-08-28:
I'm happy that it worked, again sorry for the inconvenience. I'll update the flashing guide soon. Let me know what do you think about the firmware

Nevvman18, 2024-08-28:
Make sure you have a fully clear sky visibility, I had the best results outside and without much effort it got fix in about 2 minutes. The newer GPS may have worse sensitivity so that could be the cause too. Glad to hear that it works!

Nevvman18, 2024-08-28:
Something must be wrong here, when I'll have more in stock I will try to test the GPS, but all my tests in the city town were successful and it always got the fix. If you can, try with another one, maybe this one is faulty. Also you can try to extend the GPS NMEA wait time (described in docs) to 2100ms to ensure that the GPS data gets captured. The library you are using is ok.
The current firmware doesn't support the pressure sensor because I don't own the SGP version, so for the current tests you can dismount it.

videobuff, 2024-08-28:
All is working now. I have to get used to the buttons, what they do and  when.  Question: I programmed my Horus id en get only the call sign. Is it possible to extend it with a short piece of test, for instance to get PA0ESH-testing of something else ? I don't have to send it to the amateur sondehub site. In morsecode, I get PA0ESH-TEST which I put in the call sign, and then a series of numbers. Is there any meaning to the numbers ? In the previous firmware there was the possibility to define the duration of transmissions and the silence in between. The TIME_SYNC settings. In the current firmware, the Horus starts immediately after the morse and the morse immediately after the Horus. It will drain the batteries really fast.
Thanks so far...
Erik

[image]
[image]

videobuff, 2024-08-31:
Franek, no further comments so far. I got the manuals sorted out and now know what the FW does and doesn't do. I can handle the button procedure. Programming issues i had were due to low batteries while programming so i use fresh batteries now. In the end i opted for pips and horus v2. Been testing now for some time at ground level and all ok. Sent one sonde to a friend in the west of Holland and he may launch it with an anameur balloon and then we see how it fares under real conditions.

Nevvman18, 2024-08-31:
Nice to hear that. Waiting for a response about the flight! In the next days I hope to drop an update of the fw.

closed 2024-09-03 by videobuff