Nevvman18/rs41-nfw #3: uBlox Dynamic Model Setting?

darksidelemm, 2024-09-02:
Maybe I missed something, but I couldn't spot anywhere in your code where you are setting the dynamic model of the uBlox GNSS chipset.
If you don't set this to one of the airborne dynamic models (I recommend Airborne 1G), the GNSS receiver will drop lock at 18km altitude. Not good for a high altitude balloon flight!

I'd also highly recommend using the ublox binary protocol rather than NMEA. There's plenty of examples of using this in other high altitude balloon tracker projects.

Nevvman18, 2024-09-03:
I missed that, thanks. Didn't know that the default mode doesn't support higher altitudes. In the free time I will implement that in the firmware.

Nevvman18, 2024-09-03:
Published a [new release](https://github.com/Nevvman18/rs41-nfw/releases/tag/v23) with experimental support for this function. It wasn't tested right now, because I don't have an SDR capable of spoofing the GPS for testing the higher altitudes. We will have to probably wait for a real HAB flight to confirm that it's working. The source code of the function is based on the RS41HUP, so I hope it's alright, becasue as mentioned somewhere else, the new GPS doesn't have a public, complete datasheet.

darksidelemm, 2024-09-03:
The protocol specification will match this: https://content.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf

darksidelemm, 2024-09-03:
From a brief look over that protocol specification, it looks like the configuration method for the uBlox chipsets has changed significantly with the 10-series chipsets, and the CFG-NAV5 message no longer exists.

You are instead going to have to set the CFG-NAVSPG-DYNMODEL setting (still to a value of 6=AIRBORNE1G), using the UBX-CFG-VALSET command. This is described pretty thoroughly in the protocol specification. 

I'd also highly recommend that this value then be polled using the UBX-CFG-VALGET command to ensure it has been set correctly. This means you will need to be able to receive ublox binary data, not just NMEA. The [code from RS41ng](https://github.com/mikaelnousiainen/RS41ng/tree/main/src/drivers/ubxg6010) is probably a suitable starting point for this - the overall message structure is the same, but you would need to add in new message definitions.

videobuff, 2024-09-13:
Although I am a total noob on STM programming in C, with some help I was able to monitor the output of the GPS.
I monitored it right after the command was written to set the gps in airmode 6 - with 80Km ceiling.
However, the only data i see is nema gps, as darksidelemm already indicated.
Also i found the the command to set the M10 into airmode 6 in a code snippet by Clive1 on the uBlox site.

[(https://portal.u-blox.com/s/question/0D5Oj00000MCOw8KAH/about-altitude-limit-release-setting)]

It looks different than what Franek is sending to the GPS.
Clive also has code to burn it permanently (one time only)

I hope Franek or someelse is able to ickup and get it done, because if we let it fly at the moment, we weill lose gps data from 18KM, but we probably get i back when desceding back to earth....

Code from Franek with the gps monitor at the end

```
if(ubloxGpsAirborneMode) {
    uBloxPacket msgcfgnav5;
    msgcfgnav5.header.sync1 = 0xb5;
    msgcfgnav5.header.sync2 = 0x62;
    msgcfgnav5.header.messageClass = 0x06;
    msgcfgnav5.header.messageId = 0x24;
    msgcfgnav5.header.payloadSize = sizeof(uBloxCFGNAV5Payload);

    msgcfgnav5.data.cfgnav5.mask = 0b00000001111111111;
    msgcfgnav5.data.cfgnav5.dynModel = 6;
    msgcfgnav5.data.cfgnav5.fixMode = 2;
    msgcfgnav5.data.cfgnav5.fixedAlt = 0;
    msgcfgnav5.data.cfgnav5.fixedAltVar = 10000;
    msgcfgnav5.data.cfgnav5.minElev = 5;
    msgcfgnav5.data.cfgnav5.drLimit = 0;
    msgcfgnav5.data.cfgnav5.pDop = 100;
    msgcfgnav5.data.cfgnav5.tDop = 100;
    msgcfgnav5.data.cfgnav5.pAcc = 100;
    msgcfgnav5.data.cfgnav5.tAcc = 200;
    msgcfgnav5.data.cfgnav5.staticHoldThresh = 0;
    msgcfgnav5.data.cfgnav5.dgpsTimeOut = 2;
    msgcfgnav5.data.cfgnav5.reserved2 = 0;
    msgcfgnav5.data.cfgnav5.reserved3 = 0;
    msgcfgnav5.data.cfgnav5.reserved4 = 0;

    send_ublox_packet(&msgcfgnav5);
    delay(250);
    send_ublox_packet(&msgcfgnav5);
    delay(250);
    send_ublox_packet(&msgcfgnav5);

  #ifdef DEBUG_UBLOX_WAIT // enless loop to see status of ublox 
  
  if (printed == false){
  xdataSerial.println("[info]: Testing uBlox CFG-NAVSPG-DYNMODEL).");
  xdataSerial.println("[info]: to run,set DEBUG_UBLOX_WAIT to true.");
  printed = true;
  }
  while (1) {
    
    if (gpsSerial.available()) {
      char c = gpsSerial.read();
      if ( (c == '\r') || (c == '\n')) {
        line[i] = '\0';
        DumpHex(&line, i);
        i = 0;
        if (++linecount > 60) {

              xdataSerial.println(c);
          while (1);  // Grind to a halt.  We should have all the data

        }
      } else {
        line[i++] = c;
            }
[... 52 lines cut ...]
24 47 4E 54 58 54 2C 30  31 2C 30 31 2C 30 32 2C  |  $GNTXT,01,01,02,
47 50 53 3B 47 4C 4F 3B  47 41 4C 3B 42 44 53 2A  |  GPS;GLO;GAL;BDS*
37 37                                             |  77
24 47 4E 54 58 54 36 30                           |  $GNTXT60
24 47 4E 54 58 54 2C 30  31 2C 30 31 2C 30 32 2C  |  $GNTXT,01,01,02,
31 2C 30 31 2C 30 32 2C  41 4E 54 53 54 41 54 55  |  1,01,02,ANTSTATU
53 3D 44 4F 4E 54 4B 4E  4F 57 2A 32 44           |  S=DONTKNOW*2D
24 47 4E 54 58 58 54 2C  30 31 2C 30 31 2C 30 47  |  $GNTXXT,01,01,0G
4E 53 53 2A 35 41                                 |  NSS*5A
24 47 4E 52 4D 43 2C 2C  56 2C 2C 2C 2C 2C 2C 2C  |  $GNRMC,,V,,,,,,,
2C 2C 2C 4E 2C 56 2A 33  37                       |  ,,,N,V*37
24 47 4E 56 54 47 2C 2C  2C 2C 2C 2C 2C 2C 2C 2C  |  $GNVTG,,,,,,,,,,
30 2C 30 30 2C 39 39 2E  39 39 2C 2C 2C 2C 2C 2C  |  0,00,99.99,,,,,,
2A 35 36 2E 39 39 2C 31  2A 33 33                 |  *56.99,1*33
24 47 4E 47 53 41 2C 41  2C 31 2C 2C 2C 2C 2C 2C  |  $GNGSA,A,1,,,,,,
2C 39 2C 39 39 2E 39 39  2C 33 2A 33 31           |  ,9,99.99,3*31
24 47 4E 47 53 41 2C 41  2C 31 2C 2C 2C 2C 2C 2C  |  $GNGSA,A,1,,,,,,
2C 2C 2C 2C 2C 2C 2C 39  39 2E 39 39 2C 39 39 47  |  ,,,,,,,99.99,99G
53 41 2C 41 2C 31 2C 2C  2C 2C 2C 2C 2C 2C 2C 2C  |  SA,A,1,,,,,,,,,,
2C 2C 2C 39 39 2E 39 39  2C 39 39 2E 39 39        |  ,,,99.99,99.99
24 47 50 47 53 56 2C 31  2C 31 2C 30 30 2C 30 2A  |  $GPGSV,1,1,00,0*
36 35                                             |  65
24 47 41 47 53 56 2C 31  2C 31 2C 30 30 2C 30 2A  |  $GAGSV,1,1,00,0*
37 34                                             |  74
```

Nevvman18, 2024-09-13:
I didn't have time to work on it yet (school term etc).
But if the Clive1's data packets seem alright with the GPS docs (I'll check that later), then the implementation of it in the firmware should be rather easy. The current method in the firmware is indeed wrong, because someone here mentioned that the M10 series uses different formats.

darksidelemm, 2024-09-13:
Both the code snippets above are still sending the obsolete CFG-NAV5 message, so will not work on the ublox 10. 
I've already mentioned the messages you are going to need to send above...

videobuff, 2024-09-14:
I deleted my post because darkside is correct - I did not notice that the coding was for version 8 & 9.
I contacted clive and he replied that UBX-CFG-VALSET for RAM is b5 62 06 8a 09 00 00 01 00 00 21 00 11 20 06 f2 4f 
He also modified his repository. 

uint8_t ubx_cfg_valset_dyn6[] = {                                 // Series 9 and 10
  0xB5,0x62,0x06,0x8A,0x09,0x00,                               // Header/Command/Size  UBX-CFG-VALSET (RAM)
  0x00,0x01,0x00,0x00,0x21,0x00,0x11,0x20,0x06,     // Payload data (0x20110021 CFG-NAVSPG-DYNMODEL = 6)
  0xF2,0x4F };

See also Clive1's repository at  # https://github.com/cturvey/RandomNinjaChef/blob/main/uBloxHABceiling.c

darksidelemm, 2024-09-15:
Yes, that looks a lot better - it's really important to read back the setting and verify that it's taken effect as well though...

Nevvman18, 2024-09-19:
I have finally had enough time to add the Clive's commands for M10 uBlox series to the RS41-NFW code. New release [here](https://github.com/Nevvman18/rs41-nfw/releases/tag/v231). For now, there is no reading function of the Dynamic Model setting, because I'll have to investigate how to do it (the VALGET query is helpful so that's what I'll be observing now). Currently, it sends the commands to the GPS twice, as a kind of fail-safe.

Nevvman18, 2024-09-19:
Wrote some simple code that would read the response from the GPS. The problem is that it's not similar to the format like I think it should be, but this will need some thinkering. The good news is that something is changing here because the response after sending the VALSET command is different.

videobuff, 2024-09-20:
Franek, worked with the new code in order to understand.

The output I see on my serial port is

> Received response: 75 2D 62 6C 6F 78 2E 63 6F 6D 2A 34 45 D A **24 47 4E 54 58 54 2C** 30 31 2C 30 31 2C 30 32 2C 48
>
Clive helped me out becuase i could not make sense of these codes. He replied that these are the hex codes for portions of NMEA Sentences.

> 24 47 4E 54 58 54 2C 30 31 2C -> $GNTXT,01,
>
You'll have to fish through the stream looking for B5 62 06 .. patterns
The response is likely to lag, several hundred milliseconds or more.
Hope it helps.

Nevvman18, 2024-09-20:
Thanks, that makes sense now.

darksidelemm, 2024-09-20:
Working with mixed NMEA and uBlox binary sentences is always going to be a bit messy... You may be better off just disabling all the NMEA output and working with the uBlox binary data directly. 
The driver structure from RS41ng already provides the frameworks for reading in ublox data, though you would probably need to modify the message types a little, as a few have changed with the ublox 10 series.

Nevvman18, 2024-09-20:
And thats probably the hardest part, with implementing the binary protocol. I'll look at the responses again, maybe there is a lag somewhere, maybe the code doesn't read everything. We'll see.
I'd like to stay with the NMEA, because of it's ease of use, though if it's really necessary then I'll research on implementing the UBX (either way I didn't see any issues *yet* with NMEA when using uBlox GPS modules for some time in other tracking projects).

darksidelemm, 2024-09-20:
I guess my point here is that you don't need to start from scratch. Other projects have implemented the uBlox protocol, e.g. the RS41ng implementation here: https://github.com/mikaelnousiainen/RS41ng/blob/main/src/drivers/ubxg6010/ubxg6010.c
It has all the state machine for handling incoming bytes, checksumming, etc... 
You would need to modify/add some message definitions to support the different configuration approach, and probably update a few of the other message types a little, but it's *mostly* there. 
I find the NAVPVT message quite handy, as it has everything needed for balloon telemetry all in one message: https://github.com/mikaelnousiainen/RS41ng/blob/main/src/drivers/ubxg6010/ubxg6010.c#L26

zanco, 2024-09-22:
I flew BZ-nextgen today. The balloon position / altitude appeared to work nicely, 23363 m altitude was burst level and no show of wrong postitions or altitude . 

https://grafana.v2.sondehub.org/d/HJgOZLq7k/basic?var-Payload=BZ-NEXTGEN&from=1726993343000&to=now&orgId=1&refresh=1m

Nevvman18, 2024-09-22:
I think that the VALGET query won't be as important now, because we can see that the GPS of the new model worked perfectly. I'll try to finally add it in the future, but I'm happy that the VALSET works properly on the new model.

darksidelemm, 2024-09-22:
I still think it's important to check that the mode is correct.

On a related note, the uBlox 10-series is a multi-GNSS receiver, though oddly only a maximum of 12 SVs were reported during the flight - I would have expected a higher number than this. I'm wondering if the ublox is outputting NMEA strings from other 'talkers' (e.g. GPNDA vs GPGGA) that TinyGPS might not be interpreting. A dump of serial output from one of these modules would be interesting to look at.
... of course, if you use the ublox binary NAVPVT message this should all be included in there anyway.

videobuff, 2024-09-23:
Same story here with an semi indoor location. Max no of satelites is 12
[image]

Nevvman18, 2024-09-23:
Yeah, but the number of satellites value is dependent on the NMEA messages mentioning GPS constellation, by the TinyGPS++ library. The actual coordinates are calculated on the GPS module using more of them.

darksidelemm, 2024-09-23:
This is why i was asking about what NMEA sentences are actually being sent. Often there is are 'combined' sentences beginning with $GN  (e.g. GNGGA) which provide multi-constellation information. It would be valuable to report this information if it exists.

Nevvman18, 2024-09-23:
There are combined sentences, but it would be only valuable in the number of visible satellites. It won't make a difference in navigation capabilities, though.

darksidelemm, 2024-09-23:
Sure... it would be nice to show though! 
It would also give a potential indication in the telemetry of what model RS41 PCB is in use (as the 4x2 PCBs would be limited to GPS only).

Nevvman18, 2024-09-23:
Like to send the information about the PCB model (including Horus)?
Also, what do you mean by the RSM4x2 being limited by the GPS? :)

darksidelemm, 2024-09-23:
What I mean about the RSM4x2 is that it uses a GPS-only receiver. It's only going to report up to some limited number of SVs, since it only sees the GPS constellation. It seems the maximum number is about 12 (which make sense, thats about as many GPS SVs you can see at once from a given location).

The RSM4x4 series has a multi-GNSS receiver, so the total number of SVs it can see will be higher, as it's seeing GPS+Glonass+Beidou+Galileo. If you report that complete number, it should make it a bit more obvious that the board in use is one of the RSM4x4 series.

Nevvman18, 2024-09-23:
Oh I see. Thanks

closed 2026-01-02 by Nevvman18