Nevvman18/rs41-nfw #4: rs4-nfw needs setting for delay between transmissions and pip's

zanco, 2024-09-06:
Hi,
Today I succesfully followed the instruction to progam a next generation RS41 with self compiled firmware in the Arduino IDE.  I used the stm32 programmer tool to flahs the compiled binary .

BZ-NEXTGEN starts transmitting after selecting the mode for TX -=on.

However the Horus V2 transmissions transmit without interruptions where I need to have an adjustable delay between the transmissions, and if I enable the bip's they interrupt the transmissions but also with no delay in between.
This makes it impossible to use 2 sondes at the same time. 

Can you add an adjustable delay between the horus transmissions and between the bips pleaes ?

Thanks,
Ben

Nevvman18, 2024-09-06:
You mean that the delay after each TX mode should be a separate value to configure? Sure, I can implement that. For now, there is an option called `modeChangeDelay` but the delay is the same for each mode. Somewhere in the future there will be an option to schedule the transmission with GPS time. Thanks

videobuff, 2024-09-06:
According to me it is there. It's called `int modeChangeDelay `and i have set it to 1500. (default it is 0)

Nevvman18, 2024-09-06:
Is the `modeChangeDelay` option sufficient for you or do you need the adjustable one for each?

videobuff, 2024-09-06:
For me it is sufficient. I tested yesterday for a couple of hours at ground
level and the 1500 ms is enough. At live flight I will most likely use
Horus and 5 pips.

Verzonden via Gmail Mobile

Nevvman18, 2024-09-06:
Great, and @zanco?

zanco, 2024-09-06:
Zanco is completely satisfied at this moment 

Might have helped if I had finished all documentation first 🫣

With some help I now even have the tracker starting at powering on.

Thanks!

Ben

Nevvman18, 2024-09-06:
No problem, thanks for feedback!

closed 2024-09-06 by Nevvman18

AdamM68, 2025-04-25:
comming back to this idea. Is there an option to set my own delay between transmissions ?
I managed to compile and start RS41 with APRS mode but it transmits every 5s . Too often ! I would like to set it to transmit ever few minutes.

Nevvman18, 2025-04-25:
Yes, everything is stated in the docs and code comments. Delay after each transmission mode in a cycle can be set with, for example, `aprsWait`.

AdamM68, 2025-04-25:
Ok, I found var aprsWait = 0 .... so it means that after APRS TX it goes immediately to next mode TX ?
It looks like after every kind of transmission I can add a delay. 
But I can not set a scenario like: horus erery 5 minutes, aprs every 10 minutes ?

My goal is a solar powered unit ... so I consider how to preserve power as much as possible :)

So far this is a great project for me , much because it is Arduino ! much more simple then Platformio ...for me :)

Nevvman18, 2025-04-27:
If set to 0 then yes, it goes as fast as it can to the next tx mode.
Currently there is no possibility to have a planned transmission or something like that, only in the way of a cycle.
However, for your use case you can enable both horus and Horus second transmission modes, set both with the same QRG and same 5 minute delay and then enable APRS with 0 delay. In this way, the Horus will enable two times every 5 minutes and then the APRS will transmit after the second Horus, giving you the APRS interval of 10 minutes, and of course Horus every 5 minutes.
Thanks for the kind words

AdamM68, 2025-04-27:
Thanks for the explanation.
And another question .. is rs41 going to sleep between TX ?  
I used to program esp/nRf  and have no experience in STM so I have a question : 
delay is not a sleep mode (in esp) so setting long delays between TX will I get any power savings ?

Adam