Nevvman18/rs41-nfw #6: Sensor data incorrect

zanco, 2024-09-22:
Hi, today I flew BZ-nextgen, new version RS41 , not shielded, bare PCB. (like I usually do with the older RS41 models too)

I noticed some strange behaviour in the sensor data:

[image]

reports stopped after burst because of batteryCutOffVoltage reached, but there are some strange bumps in the temperature report.

Also some releative humidity pops up in the data:

[image]

Nevvman18, 2024-09-22:
The two variables with a 'bouncing' graph depend on the heating algorithm, described in the OPERATION_MANUAL and your previous issue.
In short: You didn't configure the `refHeating`Mode` and leaved it on AUTO (if you don't use RTTY mode, please for the HAB flight switch it OFF). The temperature of the PCB started to fall below the heating threshold, which eventually activated the heating circuitry (reference heating resistors). The AUTO heating mode activates the heating for 10 minutes and then disables the heating. This phase is exactly one cycle on the graph between the drops in the temperature graph. Then, the sonde waited for the temperature to fall, which eventually happened and when the threshold was reached, the heating started again for 10 minutes. This cycle was happening during the flight and that is what you see as the bumps in the graph.
The second value you mention - humidity - isn't related to it at all. The sonde was using some default data format (I assume that you flew with Horus mode with the 4FSKTEST). The RS41-NFW firmware currently assigns to this place a value called `deviceDebugStatus`, which helps in examination of what is happening with the device. This is also described in the OPERATION_MANUAL. This value also has bumps, because it carries the heating information too.

Sorry for the inconvenience, from the next fw release the heating mode will be turned OFF by default.

Also, thank you very much for sharing the flight data with me, this is probably the first real HAB usage of this project.
Which firmware version you were running on the flight and what was the transmission mode?

I hope this helps.
Again, thanks
Franek

zanco, 2024-09-22:
Hi Franek, this certainly helps a lot, thanks for the information. This firmware contains some options which indeed have to be read in the manual and applied I know now ;-) 

My biggest error on todays flight was running 2 trackers in parallel on one set of batteries to save a bit of weight beacuse I needed to reach at least 19 km altitude. The seconde tracker (ESP32) both falied to get gps lock (I am to blame, the ceramic GPS antenna was next to the 100 mW TX antenna for the RS41) and does not run on battery voltage below 2.8 volt so my backup also did not work.

Todays flight was running 4fsk only ( flight ID 659 BZ-nextgen ) with rs41-nfw_v24 

if I did my calculations right the transmission interval was 34 seconds but the modeChangeDelay was set to 17000 ms.

Can I also opt to default the radioPweSetting to 5 instead of 7 which is overkill as soon as the balloon reaches a couple of 100 meters altitude.  What I would like (I can make another issue for that if you prefer) if it would be possible to enable a mode where , when below certain altitude, the transmission interval is set a bit larger and the power is set to setting 7 to achieve a better last fix :-)

Nevvman18, 2024-09-22:
Well, sure it is possible and it's a good idea. I'll implement that during this week.
Thanks for the feedback!

closed 2024-09-22 by Nevvman18