Nevvman18/rs41-nfw #7: Horus Binary v2 Data Field Conventions

darksidelemm, 2024-10-14:
Until I figure out a better way to handle assigning a single custom data template to multiple callsigns, the Horus binary decoder assumes that the custom field section matches the format used by RS41ng (the developer of which I collaborated with while developing the Horus Binary v2 mode).

I see you are using the custom field formats of:
```
    BinaryPacketV2.dummy1 = deviceDebugState * 100; //-32768 - 32767 int16_t
    BinaryPacketV2.dummy2 = mainTemperatureValue * 10; //-32768 - 32767 int16_t
    BinaryPacketV2.dummy3 = 0; //0 - 255 uint8_t
    BinaryPacketV2.dummy4 = 0; //0 - 65535 uint16_t
    BinaryPacketV2.dummy5 = 0; //unused in this decoding scheme
```
I also see that the temperature value in the main packet appears to be a thermistor value? (Unsure where this is from?) In the other RS41 firmwares the radio temperature is used here.

Please consider changing the custom fields to match RS41 as much as possible. This is particularly important if you are using sensor boom and pressure sensor data. The RS41ng format is:

- int16: Ascent rate (m/s) x100 (Noting that you would need to obtain this from a uBlox binary message, otherwise set it to 0)
- int16: External Temperature (˚C) x10
- uint8: Humidity (%)
- uint16: Pressure (hPa) x10

There is a free byte on the end where you can include your device debug state value.

This way, sensor data will immediately show up on sondehub and the grafana dashboards without having to apply the same new custom field entry to many new payload IDs.

Nevvman18, 2024-10-14:
I will include this in the next release, thanks for the correction. I'll place the debug value to the last byte.

Nevvman18, 2024-10-15:
RS41ng-compatible default payload format is now implemented in `v28`.
https://github.com/Nevvman18/rs41-nfw/releases/tag/v28

closed 2024-10-15 by Nevvman18