Nevvman18/rs41-nfw #8: Sensor Stalk Calibration

darksidelemm, 2024-10-14:
I see you're having a go at reading data off the sensor stalk - very good!

Stations running radiosonde_auto_rx collect the full calibration dataset from RS41s, which are uploaded to SondeHub. There is a script within the auto_rx repository which alows you to query the data for a serial from sondehub here: https://github.com/darksidelemm/radiosonde_auto_rx/blob/testing/auto_rx/utils/rs41cal.py

If you can find a way to make use of these calibration parameters, that would allow for considerably more accurate sensor measurements.

For example:
```
(venv) darkside@Marks-MacBook-Pro utils % python rs41cal.py W1330006
2024-10-15 08:22:45,586 INFO: Starting download of telemetry for serial W1330006
2024-10-15 08:22:50,067 INFO: Downloaded 173442 frames of telemetry
Telemetry data at time of subframe collection:
{'alt': 12.82622,
 'batt': 2.9,
 'burst_timer': 65535,
 'datetime': '2024-10-12T22:49:11.000000Z',
 'frame': 559,
 'frequency': 401.5005625,
 'heading': 301.85964,
 'humidity': 32.5,
 'lat': -34.95236,
 'launch_site': '94672',
 'launch_site_range_estimate': 102.17425372861796,
 'lon': 138.52043,
 'manufacturer': 'Vaisala',
 'position': '-34.95236,138.52043',
 'ref_datetime': 'GPS',
 'ref_position': 'GPS',
 'rs41_mainboard': 'RSM412',
 'rs41_mainboard_fw': '20215',
 'rs41_subframe': 'cE6AJQMAAA4AAAAAAFcxMzMwMDA2904AAFgCEgW0PKQGFIcyAAAA//8AAAEAAxYj6AMABAAHAL8CkbMABgCAO0QAgIlEAAAAAAAAPEIq6XPDXyhAPruSCTcinoY/ZPGrvb/RqTsAAAAAAAAAAAAAAAAAAAAAZFg0Qn+UoEDheSm7UpgPwF/EHkHDn2fA6WtZQjOausKO0k5Cw3sbQvhvUUPwN73DqMUSQZM9nEHrQRZDFOgWw0UojMMJSzZDT/ZKRW86f0WGkWnD8a+sQ403SEN7H8LDhxpixQAAAABU12FD9AxpwwAAAAAAAAAAAAAAAAAAAACJILrCAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAq6XPDXyhAPruSCTfzLqY/zQQBPTq5SzwAAAAAAAAAAAAAAAAAAAAAAAAAAAD////GAMQ8NEIAAAAAAM3MzD29/0u/R0mevWY2sTNbOYu3G4rxOQDgqkTwhUk8AAAAPwAAkEAAAKA/AAAAADMzMz9okS0/AACAPwAAAAAAAAAA5pZ+P5eCm7iqOSMw5BbNKbUmWqL96wIa7FE4PjMzMz8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA9n90QDs2gr/lL5g9AAEAATaeAb9/X0s+AAAAQLlopEEbCKVBAABAQP///8b////G////xv///8ZSUzQxLVNHAAAAUlNNNDEyAAAAAFcwOTYxOTQzADAwMDAwMDAwMDAAAAAAMDAwMDAwMDAAAIEjAAAaAgAC/jS1P6oAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA1cqkPV2jZTl/hyI5AAAAAAAAAAAJ/re8yJblPjGZGr8S2to+toRowWdVV0LWxarBhJ7Hwf28PkEeFkzCfLiLQbsy9EEAAAAAAAAAAAAAAAAAAAAABwABABQAyABGADwABQA8ABgBnmLVuGycB7EAPIh3AAAAAAAAAAD//5YAAAAAAB4fFQITEQAA',
 'sats': 9,
 'serial': 'W1330006',
 'snr': 13.3,
 'software_name': 'radiosonde_auto_rx',
 'software_version': '1.7.5-beta7',
 'subtype': 'RS41-SG',
 'temp': 20.5,
 'time_received': '2024-10-12T22:48:55.762490Z',
 'tx_frequency': 401.5,
 'type': 'RS41',
 'upload_time_delta': -0.847,
 'uploader_alt': 420.0,
 'uploader_antenna': '6 dBi Vertical',
 'uploader_callsign': 'VK5ARG',
 'uploader_position': '-34.81376,138.74881',
 'user-agent': 'Amazon CloudFront',
 'vel_h': 0.12285,
 'vel_v': 0.1975}
Subframe contents:
{'allowXdata': 0,
 'burstKill': 1,
 'burstKillFrames': 30600,
 'calT': 1.0517008304595947,
 'calTU': 1.2983077764511108,
 'calibU': [45.08631896972656, 5.018126964569092],
 'cpuTempSensorVoltageAt25deg': 1.415679693222046,
 'enableTestMode': 0,
 'enableTx': 1,
 'firmwareVersion': 20215,
 'flightKillFrames': -1,
 'frequency': 401.5,
 'freshBatteryCapacity': 8982,
 'heightOfFlightStart': 0,
 'intTemperatureCpu': 30,
 'intTemperatureRadio': 31,
 'killCountdown': -1,
 'lastTxPowerLevel': 0,
 'launchAltitude': 150,
 'lowBatCapacityThreshold': 135,
 'lowBatVoltageThreshold': 1.8,
 'mainboard_serial': 'W0961943',
 'mainboard_version': 'RSM412',
 'matrixBt': [-0.022460000589489937,
              0.44841599464416504,
              -0.6039000153541565,
              0.4274449944496155,
              -14.532400131225586,
              53.83340072631836,
              -21.346599578857422,
              -24.95240020751953,
              11.92113971710205,
              -51.02159881591797,
              17.46508026123047,
              30.524770736694336],
 'matrixP': [0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0],
 'matrixU': [-0.0025859999004751444,
             -2.2436718940734863,
             9.922942161560059,
             -3.619126081466675,
             54.35538101196289,
             -93.3011703491211,
             51.70561981201172,
             38.8708610534668,
             209.4373779296875,
             -378.43701171875,
             9.173255920410156,
             19.530065536499023,
             150.2574920654297,
             -150.90655517578125,
             -280.3146057128906,
             182.29310607910156,
             3247.394287109375,
             4083.652099609375,
             -233.56845092773438,
             345.3745422363281,
             200.2169952392578,
             -388.2459411621094,
             -3617.657958984375,
             0.0,
             225.84112548828125,
             -233.05059814453125,
             0.0,
             0.0,
             0.0,
             0.0,
             -93.06354522705078,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0,
             0.0],
 'minHeight4Flight': 600,
 'nfcDetectorThreshold': 0.125,
 'numSoftwareResets': 0,
 'numUbxDiscarded': 19,
 'numUbxStall': 17,
 'optionFlags': 14,
 'parameterSetupDone': 1,
 'polyT': [-0.0839565098285675, 0.005182474385946989, 0.0, 0.0, 0.0, 0.0],
 'polyTrh': [0.03149871900677681, 0.012434298172593117, 0.0, 0.0, 0.0, 0.0],
 'pressureLaunchSite': [-32767.998046875, -32767.998046875],
 'pressureSensor_serial': '00000000',
 'radioVersionCode': 6,
 'refCapHigh': 47.0,
 'refCapLow': 0.0,
 'refResistorHigh': 1100.0,
 'refResistorLow': 750.0,
 'refTemperatureTarget': 20,
 'remainingBatteryCapacity': 533,
 'serial': 'W1330006',
 'startIWDG': 0,
 'startupTxPower': 3,
 'taylorT': [-243.91079711914062, 0.18765400350093842, 8.199999683711212e-06],
 'taylorTU': [-243.91079711914062, 0.18765400350093842, 8.199999683711212e-06],
 'ubloxConfigErrors': 0,
 'ubloxHwVersionHigh': 4,
 'ubloxHwVersionLow': 7,
 'ubloxSwBuild': 45969,
 'ubloxSwVersion': 703,
 'variant': 'RS41-SG',
 'vectorBp': [0.08046499639749527,
              0.00021899999410379678,
              0.00015500000154133886],
 'xdataUartBaud': 0}
```

Nevvman18, 2024-10-14:
I'll see if these calibration parameters can be embedded into the firmware, because I don't understand them much now, but from my tests, the sensor is already quite accurate. For example at room temperatures, it was usually -0.2°C off. The boiling water test showed that it was -3°C off from the assumed boiling temperature. The worst part is the noise of the readings, which sometimes is up to 1-2°C, but this could be resolved by longer measurement time and averaging the results.

Nevvman18, 2024-12-28:
I have implemented the humidity reading too. Thought that the calibration values from Vaisala would be too difficult to implement (lots of data to copy into the firmware and simply I couldn't get everything to work here), so I used a different approach described here: https://github.com/Nevvman18/rs41-nfw/blob/main/fw%2FOPERATION_MANUAL.md#sensor-boom

But thanks again for mentioning this script, because the values from it helped much in implementing this. I also utilized the empirical temperature compensation of humidity values from radiosonde_auto_rx. Ground tests show that the readings are surprisingly good using my easier approach and seem to work in different conditions (but we need to test it in the air).

Franek