Nevvman18/rs41-nfw #9: no issue: Question, recovery of gps signal?

whallmann, 2024-10-17:
Hi
first i like to send congrats about your intension to get this new sonde rs41 hardware working. Great Job.

While the rs41ng and rs41hub had some problems with jamming gps and a full recovery on the old hardware, I made a fork 2 years ago on github.com/whallmann. I added some things, so a kind of watchdog if gps gets lost and make  reboot to get full recovery.  The gps unit fall to sleep if signal lost longer then 30 seconds and never came back if signal gets back.

Maybe on the new versions of the rs41 hardware gps unit do a better job and a watchdog is not needed further. For the old HW you need this thruly
So i ask you kindly to check the Recovery with the new ublox unit?  My way was:  wait to get a fix, then cover the gps antenna fully, wait some minutes then after lost of fix i bring the rs41 in good gps position again. unfortunelly the fix never came back.
regards
Wolf 
DF7PN

Nevvman18, 2024-10-17:
Thanks!
I didn't know about the gps fix not being recovered correctly. There is a gps reset pin connected to the uC so this is what I'll use for the reset watchdog. I'll perform tests on both old and new platforms and check this issue. 🙂

Nevvman18, 2024-10-17:
So, I tried a couple of times to reproduce the event of GPS going into sleep on all versions of sondes and it didn't happen. I tried blocking the signal with aluminium foil and placing the sondes into the corner of room, which resulted immediately in losing fix and satellite visibility. After waiting about 10 minutes I went outside again and both versions gathered the fix and satellite signals in about a minute or so.

The newer sondes have a better series of uBlox GPS chips so that's what I expect. The sondes receive the GPS signal all the time.

The older sondes however may suffer from the issue you describe, but after testing with my firmware I also couldn't reproduce this behavior. I'm nearly 100% sure that the 2 timeout values that are being sent to the GPS together with the Dynamic Model setting (implemented yesterday or so) are responsible for it. They are set to 0 meaning disabled, so the sonde didn't lose the fix and didn't timeout.

Best,
Franek

whallmann, 2024-10-18:
Ok Franek
thanks for investing this. I think this is not longer a Problem.
I will have a look to your sources how you init the ublox unit. maybe i can
extract the way and copy it to my both releases.
cu
Wolf

closed 2024-10-18 by Nevvman18