Nevvman18/rs41-nfw #11: Adding (LoRa) APRS

PC9DB, 2024-10-20:
Adding APRS protocol would be nice. Even better would be LoRa APRS but I don't know if the hardware of the RS41's is capable to do this.
LoRa APRS has a fast increasing receiver density, at least here in Europe.

Nevvman18, 2024-10-20:
The Lora APRS won't be possible on the RS41 hardware, because this mode needs a LoRa capable transmitter, which would be for example SX1278. The RS41 Si4032 transmitter doesn't support LoRa mode.

Apart from that, I think that the APRS would be the worst for HAB flights, mainly because it needs a high SNR for good reception. Even the supported RTTY gives a better performance, not even talking about the superior Horus Binary. The radiosonde would also need to operate at higher output power to reach a certain range than the previously mentioned modes, which would drain batteries significantly faster.
The APRS also uses a modulation that the Si4032 does not support from factory, and is obtained by a hack, which quickly changes tone frequencies. I heard about these transmissions being unreliable, with decoding errors and such.

The APRS mode probably won't be implemented very soon, because of more important features, unless someone would issue a pull request or write to me with an example of APRS modulation function (maybe RS41HUP / RS41ng devs could help).

Best,
Franek

PC9DB, 2024-10-20:
Hi Franek,

Thanks for your reply.
High SNR downside wouldn't be such a downside because the fast growing number of LoRa iGates. "Normal" 70cm APRS, that's a different story, not a lot of iGates on "normal" 70cm.

But if it is impossible with the RS41's hardware, then there is nothing to do with LoRa. Pity.

Regards,
Mark
PC9DB

closed 2024-10-20 by Nevvman18