Nevvman18/rs41-nfw #12: Sensor Data help?

KB8RCO, 2024-11-01:
I have several RS41-SGP (F100 chip) sondes with the daughterboard. I would like to read all sensor data and send APRS weather data directly from the Sonde.  I understand the APRS mode is not supported yet (and may not be), but my issue is that I don't seem to be able to find where the sensors are read.  I believe these are on the same SPI bus as the radio (and EEPROM where supplied)

Reading through the code of V30, I don't see anything about relative humidity or pressure, but I see it reported with BZ-NEXTGEN payload (zanco).  I see xdata stuff, but assumed those are sensors connected on the external connector.

Can you point me in the right direction?

Thank you!
Rob Giuliano
KB8RCO

KB8RCO, 2024-11-01:
Thanks for the quick reply!
If you need any testing pressure, I'd be glad to test.  I have an weather station station I could use as a reference to help understand the readings.  Fall (spring is probably better) seems to be a pretty good time to test as we get some variation in pressure readings.  Probably do the same with Humidity.

Rob

KB8RCO, 2024-11-01:
I know the issue with the Si4032 chip is the 625 Hz "granularity", but I wonder if anyone has tried decoding using "odd tone spacing" with direwolf?  In the configuration file (direwolf.conf) you can specify the tones.  
So with a setting of: MODEM 1200 1250:2500 you could use the 2 FSK values.  None-standard and probably unusable for most applications, but for simple Point-to-Point data transfer (like I would like to do), it should work.  

Could I just use the FSK (radio mode 2 or maybe GFSK 3)?
My application Configuration:
The RS-41 will be sitting on the roof of a building as a remote weather station.  The expectation is that the only receiver will be on the same building roof within a few yards, or the next building (still very close). 
  RS41 WX data packet => RTL_SDR UDP to Direwolf decode and TCP KISS to APRSIS32 => RF and IS according to connections.

Rob

Nevvman18, 2024-11-01:
I don't know anything about the APRS on the Si4032, apart from the overview of how it could be done, I'll do some research on it.

The FSK modes in the RS41-NFW are just an implementation of modulation activation in the radio IC. I didn't have any success in writing data to the buffer and sending them using FSK/GFSK.

However, the Si4032 doesn't support APRS from the factory, due to the APRS AFSK modulation. In rs41, it is achieved by a fast tone generation by changing the frequencies. In other projects the APRS mode is kind of a hack.

Also, I think that the rs41 won't be the most reliable remote weather station. I think that simply using a (quick example) BME280-kind of a sensor connected to a simple MCU (ATmega, stm32) and implementing communication via some kind of a known radio IC (worth mentioning is a superior LoRa on SX12xx chips) would be better. The sensors built into rs41 are built to measure wide ranges of parameters (temp. -100 to 200 °C, 0-1050 hPa are ranges that could be easily achieved with it) but their accuracy isn't the best. Uploading to APRS portals can be implemented with simple iGates (LoRa is used in some projects, too), unless you need an actual radio signal.

Regards,
Franek

KB8RCO, 2024-11-01:
All that is true, but this is more of an academia activity with already gathered sondes than a real project.
My job is to see what I can get working and the weather station aspect is just for "my" experimentation and learning.
Another project will (hopefully) be the regular launches of sondes on a HAB with the students experimenting with different aspects.  What I learn will be passed along and built upon.

At least that is the plan.
Thanks for all your background, information, and supportive help!

Rob

Nevvman18, 2024-12-08:
I'm coming with an update - I finally managed to get APRS operational on both versions of RS41, using my own way.
We made a test flight with it and everything was perfectly fine, the ground iGates read everything and the readings also appeared on the APRS maps.
I'll drop a new release until the end of this month, with an option to configure firmware for a WX station, of course utilizing the temperature hook.

Franek

Nevvman18, 2024-12-08:
And also another thing to mention - the precision of measurement from the temperature sensor is higher than I assumed, and during real conditions it works quite well (we encountered a faulty sonde which gave unstable readings).

videobuff, 2024-12-08:
> I'm coming with an update - I finally managed to get APRS operational on both versions of RS41, using my own way. We made a test flight with it and everything was perfectly fine, the ground iGates [...]

Franek, what iGate software are you using ?

Erik

Nevvman18, 2024-12-08:
During development I used Direwolf together with RTL-SDR, but it wasn't used as an iGate. It worked well, detected the sonde as a ham balloon and the location and comment appeared properly on the screen.
In flight we relied on the iGate infrastructure in Poland, which gave us quite good APRS reception, in total with 7 receivers hearing nearly all packets above 2km. I don't know however what is used on their stations. The transmitted APRS from this firmware should be compatible with all decoding devices.

Nevvman18, 2024-12-24:
Here are the recent updates with new features:
https://github.com/Nevvman18/rs41-nfw/releases/tag/v31
https://github.com/Nevvman18/rs41-nfw/releases/tag/v32

There is the mentioned APRS with tracking and WX formats, weather station mode, GPS operation modes and more.

Merry Christmas!

videobuff, 2024-12-25:
Thanks Franek. Loaded it up (v32) yesterday and works perfectly. Thank you for all the effort. 
Merry Christmas and the best wishes for 2025. 
Erik, PA0ESH

[image]

DJTacki, 2025-01-06:
The accuracy and stability should be very high as this is one of the core functions of the device. Here is a document about it: https://www.vaisala.com/sites/default/files/documents/White%20paper%20RS41%20Performance%20B211356EN-A.pdf

The issue is the whole boom sensor loop is analog, hence that it requires extensive calibrations. This has been done in the factory for every sonde, and hard to DIY without a weather chamber.

My method to use the boom sensors is to reverse-engineer the boom handling functions of the original fw, which already contains the correct calibrations for every sonde. Factory calibration data could be exported before erasing the original fw.

closed 2026-01-02 by Nevvman18