Nevvman18/rs41-nfw #13: NO GPS After Flashing v3

obi7zik, 2024-11-28:
Thank you for your efforts in developing the RS41NFW firmware. I successfully flashed it on my RS41 (board version 412), and everything works well except for the GPS. Despite extended time in open sky, I cannot get a GPS fix. To confirm the GPS is functional, I flashed RS41NG, and it obtained a fix within a few minutes. Any advice on resolving this issue would be greatly appreciated.

Nevvman18, 2024-11-28:
That's weird. If you can, please try it with another sonde. Also, reboot the firmware once after flashing to ensure that the GPS receives correct settings from the MCU. If you have a clear, full sky view (outside), ensure that there is no signal interference (there is some coming from the east currently).
Note, that the NFW gives a green OK light not after geting a GPS fix like RS41ng, but after reaching satellite number threshold, above 5 by default. Whether the sonde obtained the position (coordinates) or the satellite count is correct, it can be verified through for example radio transmision (Horus giving much of useful data).

obi7zik, 2024-11-29:
Thank you for the clarification. I'll try testing with another sonde if possible and ensure I reboot the firmware after flashing. The detail about the green OK light behavior in NFW versus NG is helpful—I’ll verify the satellite count and position through the radio transmission using Horus or another tool.

I’ll also double-check for interference and ensure a completely clear sky view. I'll report back with any findings after further tests. Thank you for your assistance!

videobuff, 2024-12-02:
I have just now flashed 2 sondes ( a new and an old Vaisala ) with this firmware and I can confirm that both acquire a GPS fix within minutes. The Vaisala's are indoors at the edge of the window. Note that i am always using fresh batteries.

Nevvman18, 2024-12-02:
Erik, you mean the 3.0 release, right? 
The battery note may also be helpful, thanks.

videobuff, 2024-12-02:
Yes. If you look right now at Amarwue sondehub, you can see the NFW version on line. The older version with the 3.0 firmware also came on within minutes from semi-indoors

videobuff, 2024-12-02:
[image]

Nevvman18, 2024-12-24:
New releases should be able to catch the fix faster, by reducing radio noise when searching for satellites on the ground (in my tests this helps a bit):
https://github.com/Nevvman18/rs41-nfw/releases/tag/v31
https://github.com/Nevvman18/rs41-nfw/releases/tag/v32

Merry Christmas!

closed 2024-12-24 by Nevvman18