Nevvman18/rs41-nfw #14: Alternating Horus V2 frequencies

PC9DB, 2024-12-09:
Hello,

It would be a nice feature if the software would allow for two Horus V2 frequencies to be programmed which are used both during flight.
For example every odd minute a transmission on 434.714MHz and every even minute a transmission on 437.600MHz.

By using two frequencies, the number of receiving ground stations is improved.

Thank you in advance!

Best regards,
Mark de PC9DB

Nevvman18, 2024-12-09:
Of course I can do that, thanks

PC9DB, 2024-12-11:
Thank you very much.

Best regards,
Mark PC9DB

Nevvman18, 2024-12-11:
I have already successfully implemented it, with additional repeat counter and adjustable additional delay to have a more customizable Horus telemetry. I will drop an update before the end of this month, but I have to clean up the code a little bit. This update will bring, besides that, also full APRS reporting and GPS restart watchdog in case of interference.

Franek

whallmann, 2024-12-12:
Hey Franek,

good work. These additional options will help much. Thanks

cu Wolf

Nevvman18, 2024-12-24:
Here are the new updates with the mentioned features:
https://github.com/Nevvman18/rs41-nfw/releases/tag/v31
https://github.com/Nevvman18/rs41-nfw/releases/tag/v32

Merry Christmas!

closed 2024-12-24 by Nevvman18

PC9DB, 2024-12-26:
Excellent!!! Thanks.