Nevvman18/rs41-nfw #15: Bugs

om3bc, 2025-01-11:
I tried your software. Everything seemed fine until I tried RTTY transmission. Unfortunately, there were several errors in the RTTY message generation.
1. The CRC is generated incorrectly, so Horus and DlFldigi both consider all messages as incorrect
2. The gpsTime is generated incorrectly. It is easier to use the gpsHours, gpsMinutes and gpsSeconds variables
3. The Horus program only accepts integers as altitude data
4. The temperature conversion does not work either
If you require it, I can send you the corrected script

Best regards
Joe, OM3BC

Nevvman18, 2025-01-11:
Hi,
Thanks for the observations!
Will you need the RTTY soon? I'm sorry but I'm busy at the moment, so the bugfix for RTTY won't be soon, unless it is really needed by anyone 😉.
Also, I don't quite understand what you mean by the temperature conversion. Could you please send the corrected script?

Again thanks,
Franek

om3bc, 2025-01-11:
Hi Franek,
here is the korrected script. I also added the ability to specify the RTTY
transmission mode (7N1, 7N2, 8N1, 7N1.5).

Joe

Franek ***@***.***> ezt írta (időpont: 2025. jan. 11., Szo,
11:19):

Nevvman18, 2025-01-11:
Sorry but I don't see where you have attached the script