Nevvman18/rs41-nfw #18: GPS LONG leading Zero - AGAIN

KB8RCO, 2025-02-01:
https://github.com/Nevvman18/rs41-nfw/issues/17

Well, I just changed formats from aprsOperationMode=1 to aprsOperationMode=2.
In this format, the leading zero was  '-' .  No abs() function was used so the 03d just placed the sign there.
 
There are 4 places to add the abs() function -> 2 for latDegrees and 2 for lonDegrees in the sprintf statements.
In V48 I believe they are lines 
    2489      2545  for latDegrees
    2500      2556 for lonDegrees

It appears aprsHabFormat uses the same as aprsOperationMode=1.

Sorry, I missed that the first go around.

Robert Giuliano
KB8RCO

Nevvman18, 2025-02-01:
Oh. Sorry about that, I missed the second mode. Today I'll release a fix.

Franek

Nevvman18, 2025-02-01:
Done in [v49](https://github.com/Nevvman18/rs41-nfw/releases/tag/v49), thanks.

closed 2025-02-01 by Nevvman18