Nevvman18/rs41-nfw #20: defaultModeChangeDelay not possible to set in code.

videobuff, 2025-02-02:
Although you mention defaultModeChangeDelay at line 146

`unsigned long horusSecondTransmissionInterval = 0;  //set to 0 for default delay (defined in defaultModeChangeDelay), otherwise will deterimne delay between first and second Horus transmission
`

it is nowehere defined in the code.

Nevvman18, 2025-02-02:
Oh, sorry, I had to overlook it. A few releases ago there was a change in delay handling. Now each transmission has an independent delay, and the horusSecondTransmissionInterval is used to delay transmissions of 4fsk between each if used with the horusSecondTransmissionRepeatCount. If the repeat count is set to 0, then this delay is just the one between the standard Horus transmission and this horusSecondTransmission.
I hope that this describes it properly 🙂, I'll modify the comment in the next release

Franek

videobuff, 2025-02-02:
All clear. I have been off line due to private reasons for a while so I picked up at rev 49 and had it missed out. Better the current way. Keep upmthe good work Franek.
73's
Erik

closed 2025-02-02 by videobuff