Nevvman18/rs41-nfw #21: Compilation Error

dl1esk, 2025-03-09:
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino: In function 'void initGPS()':
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:31:23: error: 'PC7' was not declared in this scope; did you mean 'PC_7'?
   31 | #define GREEN_LED_PIN PC7  //green led - reversed operation, pin HIGH=led_off, pin LOW=led_on!
      |                       ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1442:16: note: in expansion of macro 'GREEN_LED_PIN'
 1442 |   digitalWrite(GREEN_LED_PIN, LOW);
      |                ^~~~~~~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino: In function 'void selectReferencesHeater(int)':
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:41:18: error: 'PC6' was not declared in this scope; did you mean 'PC_6'?
   41 | #define HEAT_REF PC6  //reference heating resistors - LOW=disabled, HIGH=enabled - draw about 180mA of current and after a while may burn skin
      |                  ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1491:22: note: in expansion of macro 'HEAT_REF'
 1491 |         digitalWrite(HEAT_REF, LOW); //disable reference heater
      |                      ^~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:41:18: error: 'PC6' was not declared in this scope; did you mean 'PC_6'?
   41 | #define HEAT_REF PC6  //reference heating resistors - LOW=disabled, HIGH=enabled - draw about 180mA of current and after a while may burn skin
      |                  ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1502:22: note: in expansion of macro 'HEAT_REF'
 1502 |         digitalWrite(HEAT_REF, HIGH);
      |                      ^~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:41:18: error: 'PC6' was not declared in this scope; did you mean 'PC_6'?
   41 | #define HEAT_REF PC6  //reference heating resistors - LOW=disabled, HIGH=enabled - draw about 180mA of current and after a while may burn skin
      |                  ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1513:22: note: in expansion of macro 'HEAT_REF'
 1513 |         digitalWrite(HEAT_REF, HIGH);
      |                      ^~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:41:18: error: 'PC6' was not declared in this scope; did you mean 'PC_6'?
   41 | #define HEAT_REF PC6  //reference heating resistors - LOW=disabled, HIGH=enabled - draw about 180mA of current and after a while may burn skin
      |                  ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1524:22: note: in expansion of macro 'HEAT_REF'
 1524 |         digitalWrite(HEAT_REF, HIGH);
      |                      ^~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino: In function 'void selectSensorBoom(int, int)':
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:50:15: error: 'PC10' was not declared in this scope; did you mean 'PC_10'?
   50 | #define SPDT1 PC10
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1826:20: note: in expansion of macro 'SPDT1'
 1826 |       digitalWrite(SPDT1, powerState);
      |                    ^~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:51:15: error: 'PC11' was not declared in this scope; did you mean 'PC_11'?
   51 | #define SPDT2 PC11
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1827:20: note: in expansion of macro 'SPDT2'
 1827 |       digitalWrite(SPDT2, powerState);
      |                    ^~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:52:15: error: 'PC12' was not declared in this scope; did you mean 'PC_2'?
   52 | #define SPDT3 PC12
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:1828:20: note: in expansion of macro 'SPDT3'
 1828 |       digitalWrite(SPDT3, powerState);
      |                    ^~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino: In function 'void modeChangeDelayCallback(long unsigned int)':
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:31:23: error: 'PC7' was not declared in this scope; did you mean 'PC_7'?
   31 | #define GREEN_LED_PIN PC7  //green led - reversed operation, pin HIGH=led_off, pin LOW=led_on!
      |                       ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:2118:24: note: in expansion of macro 'GREEN_LED_PIN'
 2118 |           digitalWrite(GREEN_LED_PIN, HIGH);
      |                        ^~~~~~~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino: In function 'void redLed()':
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:30:21: error: 'PC8' was not declared in this scope; did you mean 'PC_8'?
   30 | #define RED_LED_PIN PC8    //red led - reversed operation, pin HIGH=led_off, pin LOW=led_on!
      |                     ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:2689:16: note: in expansion of macro 'RED_LED_PIN'
 2689 |   digitalWrite(RED_LED_PIN, LOW);
      |                ^~~~~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:31:23: error: 'PC7' was not declared in this scope; did you mean 'PC_7'?
   31 | #define GREEN_LED_PIN PC7  //green led - reversed operation, pin HIGH=led_off, pin LOW=led_on!
      |                       ^~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:2690:16: note: in expansion of macro 'GREEN_LED_PIN'
 2690 |   digitalWrite(GREEN_LED_PIN, HIGH);
[... 58 lines cut ...]
      |             ^~~~~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:50:15: error: 'PC10' was not declared in this scope; did you mean 'PC_10'?
   50 | #define SPDT1 PC10
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:3938:11: note: in expansion of macro 'SPDT1'
 3938 |   pinMode(SPDT1, OUTPUT);
      |           ^~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:51:15: error: 'PC11' was not declared in this scope; did you mean 'PC_11'?
   51 | #define SPDT2 PC11
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:3939:11: note: in expansion of macro 'SPDT2'
 3939 |   pinMode(SPDT2, OUTPUT);
      |           ^~~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:52:15: error: 'PC12' was not declared in this scope; did you mean 'PC_2'?
   52 | #define SPDT3 PC12
      |               ^~~~
C:\Users\Basti\Downloads\rs41-nfw_v55\rs41-nfw_v55\rs41-nfw_v55.ino:3940:11: note: in expansion of macro 'SPDT3'
 3940 |   pinMode(SPDT3, OUTPUT);
      |           ^~~~~

exit status 1

Compilation error: 'PC7' was not declared in this scope; did you mean 'PC_7'?

any Ideas?

Nevvman18, 2025-03-09:
Looks like an inconsistency in configurations.

You either have an old MCU selected in the compiler (F1 series) together with the `#define RSM4x4` (being the new sonde PCB version), or vice versa, with L4 (newer MCU) selected and RSM4x2 (old board) set.

Check the settings and let me know if it is alright.

Nevvman18, 2025-03-09:
Judging by the logs, in the compiler settings you have F1 selected, but the firmware file has a model setting configured as RSM4x4, that's why it says that some of the pins aren't declared here.

dl1esk, 2025-03-09:
yes was my fault ..selected the old Board and erverything went fine :-)

Nevvman18, 2025-03-09:
Great to hear it!