Nevvman18/rs41-nfw #23: CW modulation

iz2fly, 2025-04-06:
I have already tried some firmware for RS41 in CW mode but all have the same problem.
There is a big phase noise problem that disturbs at 100-150 KHz above and below the transmission frequency.
It is probably a problem of the Si4032 or a hardware problem of the RS41. I do not know if it can be solved by modifying the firmware.

https://iz2fly.altervista.org/radiosonde/

Nevvman18, 2025-04-06:
I didn't observe this issue myself and it's the first time I've heard about it. 
However I'm nearly 100% sure that it's a Si4032 design flaw or something, and can't be improved with firmware.

iz2fly, 2025-04-06:
Thanks for the reply. 
I was hoping it was a firmware problem but you confirmed my doubts.
I also tried with your firmware but nothing changes.
I tried with an ADF4351 chip the signal improves a lot.
So the problem is the Si4032.
I will have to change the beacons because they disturb the OM reception a few kilometers away.
In APRS mode it works very well with the firmware modified by my friend I2NOS, you can find it on my website.

closed 2026-01-02 by Nevvman18