Nevvman18/rs41-nfw #24: Succesful flight

videobuff, 2025-04-16:
Franek,
Today I had for the first time balloons fly from my qth. And even two of them.
One setup fo a burst flight and one a floater. 
Modes enabled were pip, Horus & aprs, and the other only Horus & aprs.
I made a mistake programming both with the same Horus and aprs id’s so it was a bit of confusion but since the first one landed in the North Sea and the other one is still flying I call it a successful exercise.
They flew as horus PA0ESH-NFW and aprs PA0ESH-12
Thanks again for all the work you did. I have 10 more sondes laying around so plenty to fly.
We tried the burst for above 20 km but for one or another reason there was too little gas in the balloon, and it stopped at 16 km.

73's
Erik

[https://www.youtube.com/watch?v=v9WIs4gPaks](https://www.youtube.com/watch?v=v9WIs4gPaks)

AdamM68, 2025-04-19:
is floater still alive ?

videobuff, 2025-04-19:
No, both ended up in the Northsea. Got the lift wrong.

Nevvman18, 2025-04-21:
Hi Erik,
Sorry for the delay in response, I was a bit busy lately.
It's so great to hear from you about the flight! I saw it on SondeHub and it looked very nice!
Thanks for testing the NFW

Best,
Franek