Nevvman18/rs41-nfw #27: Fluctations in external temperature measurements

maxkup14, 2025-06-13:
Hi Franek, 

Wonderful firmware, really nice job! However, while testing your firmware on two RS41-SGP (V1641331 and V1840353) at the same time, I noticed a clear difference in the fluctuation behavior of the "ext_temperature" values. 

- In the first dataset, the external temperature shows relatively stable values with minor variations. The GPS module was turned off (gpsOperationMode = 0)

[image]

- The second dataset shows bigger fluctuations within the same sample window. In this case, the GPS was turned on (gpsOperationMode = 1)

[image]

I also noticed that when GPS is ON, the interval between horus binary transmissions is bigger (3-5 seconds longer) than what I have set in the code.
Both of the probes were set on the same parameters (except gpsOperationMode), see below :

```
//===== Interfaces
//SPI_2 interface class (radio communication etc.)
SPIClass SPI_2(PB15, PB14, PB13);  // MOSI, MISO, SCK for SPI2
// ublox gps              rx    tx
HardwareSerial gpsSerial(PA10, PA9);
int gpsBaudRate = 9600;

#else
#error "Please define the pcb model!"
#endif

// XDATA (2,3)          rx    tx
HardwareSerial xdataSerial(PB11, PB10);

//===== Radio signals config
bool radioEnablePA = true;          //default tx state

bool pipEnable = false;             //pip tx mode
float pipFrequencyMhz = 434.2;      //pip tx frequency
unsigned long pipWait = 1;           //delay after pip transmission, SET TO 0 IF PIP NOT USED, in milliseconds. If the firmware cant keep up with the delay, it transmits as fast as possible, else this is the delay after pip and before horus. 0 - disable, 0<delay<5000 - standard delay, 5000<delay - delay + radio sleep mode and recalibration (if radioStableWait enabled) and additional flight computing and gps receiving.
int pipLengthMs = 1000;             //pip signal length in ms
int pipRepeat = 3;                  //pip signal repeat count in 1 transmit group
int pipRadioPower = 4; //TX power, 0 = -1dBm (~0.8mW), 1 = 2dBm (~1.6mW), 2 = 5dBm (~3 mW), 3 = 8dBm (~6 mW), 4 = 11dBm (~12 mW), 5 = 14dBm (25 mW), 6 = 17dBm (50 mW), 7 = 20dBm (100 mW)

bool horusEnable = true;  //horus v2 tx mode
float horusFrequencyMhz = 434.3;
unsigned long horusWait = 5000;           //same as in pip but after horus
unsigned int horusPayloadId = 811;
int horusBdr = 100;
int horusRadioPower = 4; //TX power, 0 = -1dBm (~0.8mW), 1 = 2dBm (~1.6mW), 2 = 5dBm (~3 mW), 3 = 8dBm (~6 mW), 4 = 11dBm (~12 mW), 5 = 14dBm (25 mW), 6 = 17dBm (50 mW), 7 = 20dBm (100 mW)

bool horusEnableSecondTransmission = false;  //enable second horus transmission, may be used for example to transmit on different frequencies or at different intervals
float horusSecondTransmissionFrequencyMhz = 434.2;
unsigned long horusSecondTransmissionWait = 100;           //same
unsigned int horusSecondTransmissionRepeatCount = 5000;
unsigned long horusSecondTransmissionInterval = 1000;  //set to 0 for default delay (defined in horusWait or powerSaveHorusWait), otherwise will deterimne delay between transmissions if used with repeatCount > 1 (more than 1 teransmission in cycle)
int horusSecondRadioPower = 4;

bool aprsEnable = false;
float aprsFrequencyMhz = 448.07;
unsigned long aprsWait = 100;
char aprsCall[] = "mk1359";  // Callsign
String aprsComment = " @RS41-NFW"; //Leaving the nfw comment would help identify the usage of this firmware :)
char aprsSsid = 11;            // SSID for the call sign
char aprsDest[] = "APZNFW";    // Destination address for APRS
char aprsDigi[] = "WIDE2";     // Digipeater callsign
char aprsDigiSsid = 1;         // Digipeater SSID
char aprsSymbolOverlay = 'O';  // Symbol overlay
char aprsSymTable = 'a';       // Symbol table (e.g., 'a' for standard symbol)
int aprsOperationMode = 1;     //1 - standard telemetry format (similiar to RS41HUP), 2 - WX format (weather station)
int aprsRadioPower = 7; //TX power, 0 = -1dBm (~0.8mW), 1 = 2dBm (~1.6mW), 2 = 5dBm (~3 mW), 3 = 8dBm (~6 mW), 4 = 11dBm (~12 mW), 5 = 14dBm (25 mW), 6 = 17dBm (50 mW), 7 = 20dBm (100 mW)

#define CALLSIGN "mk1359"         //used for morse and rtty
bool rttyEnable = false;          //rtty tx mode, compliant with UKHAS format
[... 97 lines cut ...]
//lowAltitudeFastTx mode
int lowAltitudeFastTxThreshold = 0;             //set to 0 to disable. When sonde is descending after a burst, when it goes below this threshold, it goes into a 'lowAltitudeFastTx' mode, in which it only transmits horus packets as fast as it can, to possibly catch the lowest frame, works only with horus (APRS would overload the infrastructure)
unsigned long lowAltitudeFastTxDuration = 480000;  //duration of how long this mode will work, in milliseconds (default 480000ms = 8minutes)
int lowAltitudeFastTxInterval = 1;                 //delay in ms between transmissions in this mode, should be left at 1 to catch the lowest frame possible

//Flight computing
unsigned int flightDetectionAltitude = 600;  //default flight detection altitude in meters (if exceeded, the sonde knows that the flight began)
unsigned int burstDetectionThreshold = 100;  //describes threshold value, which if exceeded (below maxAlt) deterimnes if the balloon has burst (2000m seems reasonable, due to some being floaters or getting 'unsealed')

//System
bool autoResetEnable = true;                              //automatically reset the CPU after specified time below, useful in stationary continuous use, to prevent from overflowing some variables
#define SYSTEM_RESET_PERIOD 28800000  // 8 hours in milliseconds
int buttonMode = 1;                                 //0 - button operation disabled, 1 - the button can turn the sonde OFF, 2 - extended mode, allowing to control radio power and if the transmission should be enabled and shutdown. If you want to fly a sonde with PV or on 1xAA hardware, consider disabling the button and shorting its pins for always closed state. NOTE: the button will only operate when the sonde has fully started (after calibration, compensation, hardware configuration etc.)

//dataRecoder config
bool dataRecorderEnable = false;              //enables mode, in which the sonde transmits some recorded and debug data to the ground via additional APRS comments (described in repo). Format: ...NFW;[maxAlt];[maxSpeed];[maxAscentRate];[maxDescentRate];[maxMainTemperature];[minMainTemperature];[maxInternalTemp];[minInternalTemp];[ledsEnable];[healthStatus];[gpsResetCounter];[beganFlying];[burstDetected];[isReferenceHeaterOn];[radioPwrSetting];[currentGPSPowerMode];[radioTemp];...
unsigned int dataRecorderInterval = 600000;  //10 minutes by default (600000 milliseconds)
bool dataRecorderFlightNoiseFiltering = true; //filter out noisy data on ground and duing position gathering and include in the measurements only flight data
```

I have also tested the newest variant (RS41-SGP X0324487, PCB variant RSM425), once with GPS off and once with GPS on, the behavior of the sensor was the same as previously mentioned. My question is : How can the GPS module be causing such a difference in measurements ?

Thx, keep is going ! I am not a developer myself, but I do have access to several probes from this new X series and would be happy to assist with testing or to provide sample data if that would be helpful.

Max (mk1359)

maxkup14, 2025-07-15:
Hi Franek,

Thanks for your reply, no worries!

Just to clarify: I'm consistently using matched PCB and sensor boom pairs (i.e., no mixing between different sondes). So far, I haven't noticed any differences across different PCB versions when running the latest firmware, including the newer RSM425 (X series). The fluctuation issue with GPS enabled appears consistent across all variants.

Interestingly, I once configured the RS41 for stationary use without GPS and left it outside overnight. The temperature readings were stable and matched well with my home weather station, which led me to suspect that the GPS might be contributing to the observed instability.

Let me know if there’s any way I could help with testing or provide sample data. I’d be happy to contribute if it can support further development.

Max

Nevvman18, 2025-07-15:
Wow, thanks for the thorough writeups and time spent on working with firmware!

I have an idea which could work here, as soon as I get more time I'll look deep into that. I'll reach you back as soon as I find the possible solution or when I'll need some help.

Again, big thanks 😁
Franek

Nevvman18, 2025-07-27:
Today I tried to search for a solution. If the sensors are read in between the GPS' 1 second data transmit interval (in the half of it) the noise could have lowered, but the implementation is still bad, so no great news so far. Definitely a thing to investigate

Nevvman18, 2026-01-17:
`v62` release fixes this issue, measurement noise is absent now.

Franek