Nevvman18/rs41-nfw #32: CATS support?

geokscott, 2025-08-30:
Have you considered adding CATS support?

geokscott, 2025-09-22:
> Nevvman18 left a comment [(Nevvman18/rs41-nfw#32)](https://github.com/Nevvman18/rs41-nfw/issues/32#issuecomment-3317703860)
>
> Hi, I haven't looked into this protocol yet and didn't feel the need to do so.
> Is it a protocol that is worth implementing? I'm not really familiar with it. If so, then I'll look into the docs and try to add it.
>
> Thanks,
> Franek

You can read about it here: https://cats.radio/
CATS is the next generation of APRS, it's much better.
Nevvman18 implemented it in his firmware: https://github.com/Nevvman18/rs41-nfw
I would like to deploy it in my RS41's, but yours is the only implementation so far that works on the new RS41's

Thanks - George

Nevvman18, 2026-01-18:
Hey George, 
After a long time I came to the conclusion that in the near future I probably won't work on the CATS modem. I had never used it and it seems rather unpopular in my opinion. It seems unfortunate, because it may be better than the APRS. Additionally, the RSM4x2 NFW release has only 2% memory left, which means that even the smallest piece of code will not fit there anymore. I'm sorry for it and I hope you understand the decision.

Best regards,
Franek

geokscott, 2026-01-18:
I totally understand. Consider this though. Yes CATS is better than APRS. APRS is too long in the tooth for it's own good. If we all don't support new changes, it will slow progress or progress will die on the wine. Memory is a problem, I'd suggest creating a parallel build that cuts out APRS and replaces with CATS. If we all don't support new technologies, we would never make new progress.

Thanks - George

AdamM68, 2026-01-18:
CATS  should add support to esp32 ... so far there is only the hardware for an stm32-based mobile CATS transceiver.

AdamM68, 2026-01-19:
> Why ESP32? 

... because there is plenty of low cost hw ( esp32+lora  ) and much easier to program (Arduino, VSC ) 
.. not to mention nRf52+lora :)   .. in my opinion better then stm 

In my opinion if CATS want to be next APRS it should let people use esp32 and nRf devices as well as stm.

geokscott, 2026-01-19:
The point is, no one is stopping you or anyone else from developing on the ESP platform. It's totally open source, so go for it. It will possibly make adoption go faster!

PS: I have developed on STM32 with Arduino UI and VSC, it's very easy.